nasa-jpl/osr-rover-code #5: Code is only compatible with Python 2. It should (also?) be made to run in Python 3.

masasin, 2018-08-03:
It's not very difficult, and I might be able to do it if there are no takers. Python 2 will EOL on 1 January 2020.

asasine, 2018-08-28:
What specifically about this code is only compatible with Python 2?

masasin, 2018-08-28:
e.g., print statements instead of print functions. I didn't look much deeper.

Things that often trip people up are xrange or iteritems etc. If you create a list that is reused (e.g., with range), it'll be consumed in py3 since it's a generator.

vssystemluba, 2018-08-28:
I also know that the roboclaw library we're using to easily communicate with the roboclaw motor controllers was written for Python 2.  I'm not sure if a Python 3 version exists or if their current version would be compatible, but I'm guessing the heavy lifting to make the whole thing python 3 compatible might be on that side.

asasine, 2018-08-28:
Are you referring to _roboclaw.py_ or its dependency `serial`? I did some short digging and found a pip package _pyserial_ that appears to be the same package. It's hard to tell without a dependency/package list for the source.

pabloriveracelerity, 2018-08-31:
I would only target Python 3 at this point because the target hardware is ever evolving forward (the RPi).

I'd say Python 3.4 should be a good target for the time being.

masasin, 2018-08-31:
I'd suggested 3.6+ instead, since things were introduced that make coding more convenient, as well as better asyncio support.

pabloriveracelerity, 2018-09-01:
Note that numpy is used in the project and it seems to only support 3.4.x

masasin, 2018-09-01:
That information is incorrect. It supports 3.4+, which includes 3.7. I use
it.

capsulecorplab, 2018-09-04:
numpy, scipy, and a number of other projects in the python developer community are beginning to drop support for legacy python (2.7), as it's been slowing down their ability to develop new and better features for these packages.

pabloriveracelerity, 2018-09-05:
Quick note:

Ive started moving things to Python 3.6+ with the unit tests.

pabloriveracelerity, 2018-09-13:
Now that I have a basic hardware setup, I will test installing the software and report any issues from upgrading to Python 3.6+

Achllle, 2020-11-15:
closing because we're using ROS now and we want to support Kinetic + Melodic which are (mostly) py2.7

closed 2020-11-15 by Achllle