nasa-jpl/osr-rover-code #8: Add package dependency list for ease of getting started

asasine, 2018-08-28:
The source has some pip dependencies which aren't explicitly documented.

Using a _requirements.txt_ or a workflow like [Pipenv](https://pypi.org/project/pipenv/) would aid developers with getting started. Note that Pipenv is the [recommended tool](https://packaging.python.org/guides/tool-recommendations/#application-dependency-management) by the Python for dependency management.

asasine, 2018-08-31:
Because the dependencies for this project are not limited to pip (see section 1.2.1 of [Software Steps](https://github.com/nasa-jpl/open-source-rover/blob/master/Software/Software%20Steps.pdf) in main repository), this likely cannot be solved using only pip or its flavors.

pabloriveracelerity, 2018-08-31:
A requirements file could be added with a note on the README that mentions the additional packages. It's still a good idea to have one.

pabloriveracelerity, 2018-09-13:
We could probably close this one for the time being and then open a new issue for pipenv integration. Given that using pipenv would require adjusting the documentation. Thoughts?

Achllle, 2020-03-07:
I would agree on closing. Since this is a ROS project, pip requirements do not make sense and instead dependencies should go in `package.xml`. The recommended install tool before actual bloom release would be rosdep.

Achllle, 2020-03-30:
@ericjunkins this can be closed. We will track dependencies in package.xml which currently only is pyserial, which was added in #43

closed 2020-07-22 by Achllle