nasa-jpl/osr-rover-code #9: Housekeeping suggestions for a contributor-friendly repo

capsulecorplab, 2018-08-30:
In addition to a `requirements.txt` as mentioned in #8, here are some other suggestions for creating a contributor-friendly python repo:
- [ ] Add a `tests/` folder containing unit tests - this can be useful for promoting a TDD workflow, as it gives contributors a metric for what the code should do without actually having to open up the source code. Personally, I like [pytest](https://docs.pytest.org/en/latest/goodpractices.html), but there are [other options](https://docs.python-guide.org/writing/tests/).
- [ ] Enable continuous integration, ci, s.a., [travis-ci](https://travis-ci.com/) or [circle-ci](https://circleci.com/), with a config file containing scripts for running unit tests - helps automate PR reviews, as the ci can check to ensure the suggested code changes haven't broken core functionality.
- [ ] Add a `CONTRIBUTING.md` - helps promote best practices for working on an issue or submitting a PR, sometimes occompanied by a code of conduct. (This one's also still on my backlog of things to do)
- [ ] Add pep8 linter to ci config - optional, but helps keep code readable.
- [ ] Enable code coverage s.a., [codecov](https://codecov.io/) or [coveralls](https://coveralls.io/) - optional, but serves as an additional confidence metric in source code.

Feel free to use this project as a reference/template https://github.com/spacedecentral/SysML.py :slightly_smiling_face:

asasine, 2018-08-31:
@pabloriveracelerity had mentioned in a [comment](https://github.com/nasa-jpl/osr-rover-code/pull/7#discussion_r213530534) in PR #7 that he is in the process of adding unit tests. I think they would go a long way towards ensuring code consistency.

Achllle, 2020-03-07:
Some other suggestions to add to this list:

- [ ] ROS only code in a ROS repo (see also #44)
- [ ] change instructions to not have the 'clone then move code', but instead clone directly into `catkin_ws/src` so that changes can be directly contributed back. Alternatively symlinks would work here as well
- [ ] have an install script (or perhaps a docker container but might be overkill) which helps setting up Travis as well

Achllle, 2020-11-15:
I think we're mostly there at this point. Honestly CI doesn't make sense for a repo like this as it hinders contributor friendliness and slows down development by adding the requirement for test coverage. For now linting and code style will be enforced through the PR process, but open to suggestions!

closed 2020-11-15 by Achllle