nasa-jpl/osr-rover-code #12: Unit tests

pabloriveracelerity, 2018-08-31:
Add unit tests to the code using the `unittest` library included with Python.

First round of tests will focus on testing the modules and not their implementation (a baseline to make sure the basics work as they should).

pabloriveracelerity, 2018-09-05:
Most of the scripts need the hardware to be connected in order to be tested. This is an area of unit tests and robotics that Ive done before and am working out a clear plan to implement. Currently working on ordering the hardware in order to have a proper testing setup.

Also note that given the way the software is written, there are not a lot of tests that would be possible without the actual hardware. Ill continue to write tests for the things that dont require hardware and then merge everything in when my hardware is setup.

ericjunkins, 2018-09-13:
The unittesting would predominately be based around hardware yes. Some of the unittests I was writing while initially developing the rover were to check communication to the motor controllers, and to make sure that given simple motor commands the encoders responded as expected

pabloriveracelerity, 2018-09-13:
Hi Eric,

Thanks for replying. I've managed to setup an Rpi for the unit tests and will continue to focus on that. 
We should have some good progress soon.

capsulecorplab, 2018-09-14:
For software contributors who don't have access to the hardware, would it be possible to "mock" hardware behavior for unit testing purposes?

pabloriveracelerity, 2018-09-14:
@capsulecorplab Note that when I mention hardware, I'm referring to the Rpi board itself. Plan is to mock the other hardware parts whenever possible to avoid having a working robot as a hard requirement for testing. Anything that can be mocked or emulated will be.

Having the tests depend on the Rpi is also a good baseline to avoid missing hardware issues. There is also an option to run the RPi OS on a VM. There is a possibility that this could be done to work on vagrant and thus we could distribute a ready to use environment that requires no purchase. Loner term goals, though. :)

pabloriveracelerity, 2018-11-10:
Some basic (unfinished) set of mocks have been added: https://github.com/pabloriveracelerity/osr-rover-code/blob/pr/issue-12-init-tests/rover/tests/test_connections.py

Have to temporarily refocus on some other code. Im confident anyone can take the mocks and continue adding tests.

Achllle, 2021-02-14:
I'm going to close this issue. With the overhauls of the code base in the past year unit testing has become unnecessary. The code will print relevant error messages to help debug the issue. Unit tests would also slow down new development, especially for people new to software.

closed 2021-02-14 by Achllle