nasa-jpl/osr-rover-code #23: Wrong URL for git-clone of ROS code

JHPHELAN, 2019-04-09:


vssystemluba, 2019-04-09:
Can you provide more details?  Where is the URL wrong?

JHPHELAN, 2019-04-09:
In the README.md instructions at https://github.com/nasa-jpl/osr-rover-code/tree/osr-ROS
after installing ROS on the RPi, creating and moving to the ~/osr_ws/src directory it says to 
     git clone https://github.com/nasa-jpl/osr-rover-code/tree/osr-ROS/ROS
however this creates a directory/file not found error. 
Clicking the green Clone or download button and saving to clipboard produces this url, which works:
     https://github.com/nasa-jpl/osr-rover-code.git
so it should be -
     git clone https://github.com/nasa-jpl/osr-rover-code.git

Also, in the initial ROS/RPi installation instructions in:
     http://wiki.ros.org/ROSberryPi/Installing%20ROS%20Kinetic%20on%20the%20Raspberry%20Pi
if you go so far as section 4.1 Updating the Workspace which refers you to Ubuntu source install instructions:
     http://wiki.ros.org/kinetic/Installation/Source
they refer to the "desktop-full" variant.  Our instructions call for the "ros_comm" variant so 
replace "desktop-full" with "ros_comm" as in -
     $ mv -i kinetic-desktop-full-wet.rosinstall kinetic-desktop-full-wet.rosinstall.oldomm   becomes
     $ mv -i kinetic-ros_comm-wet.rosinstall kinetic-ros_comm-wet.rosinstall.old
note where hyphen "-" vs underscore "_"

Jim Phelan
Mars Rover project lead
Houston Robotics / now renamed USAi Labs

ireasor, 2019-05-18:
You'll actually need to run:

$ git clone https://github.com/nasa-jpl/osr-rover-code.git
$ git checkout osr-ROS
$ git pull

This will get you the source code from the correct branch.  Just checking out osr-rover-code.git will get you the master branch.  I ran into the same issue when trying to get the Arduino code.  The README instructs you to run:

$ git clone https://github.com/nasa-jpl/osr-rover-code/tree/osr-ROS/Arduino OsrScreen

which results in: "fatal: repository 'https://github.com/nasa-jpl/osr-rover-code/tree/osr-ROS/Arduino/' not found"

I was able to download the code by checking out the repository and selecting the correct branch.

I'm unclear on what you are referring to in the ROS/RPi installation instructions, but it sounds like this may be a different issue.  It would probably be better to track this as such?

JHPHELAN, 2019-05-31:
@ireasor 
Step 1 runs ok:
pi@Fidelity:~/osr_ws/src $ git clone https://github.com/nasa-jpl/osr-rover-code.git
Cloning into 'osr-rover-code'...
remote: Enumerating objects: 206, done.
remote: Total 206 (delta 0), reused 0 (delta 0), pack-reused 206
Receiving objects: 100% (206/206), 206.63 KiB | 0 bytes/s, done.
Resolving deltas: 100% (67/67), done.
When I do step 2 above I get:
pi@Fidelity:~/osr_ws/src $ git checkout osr-ROS
fatal: Not a git repository (or any of the parent directories): .gitn
Now what?  

I also got the same error from the original instructions:
pi@Fidelity:~/osr_ws/src $ git clone https://github.com/nasa-jpl/osr-rover-code/tree/osr-ROS/ROS
Cloning into 'ROS'...
fatal: repository 'https://github.com/nasa-jpl/osr-rover-code/tree/osr-ROS/ROS/' not found

Once the ROS code is eventually installed, what do I do so I can push which buttons on the Xbox controller to see the wheels move?

JHP

JHPHELAN, 2019-05-31:
There's clearly something missing without the above steps working.  
Just for fun I thought I'd try to run robot.py
with both python and python3 [which are we supposed to use?] and get the following error:

**pi@Fidelity:~/osr_ws/src/osr-rover-code/rover $** python robot.py
Traceback (most recent call last):
  File "robot.py", line 4, in <module>
    from motor_controller import Motor
  File "/home/pi/osr_ws/src/osr-rover-code/rover/motor_controller.py", line 3, in <module>
    from roboclaw import Roboclaw
  File "/home/pi/osr_ws/src/osr-rover-code/rover/roboclaw.py", line 2, in <module>
    import serial
ImportError: No module named serial

JHP

JHPHELAN, 2019-06-08:
@ireasor  With the help of fellow USAi Labs member Charles, we were able to figure to
	cd ~/osr_ws/src/osr-rover-code
then do either 
     git checkout osr-ROS
 or 
     git fetch
then
     git pull

But the next step (way down the documentation after the explanation of the various components):
"Running		
There is a roslaunch file that will start all of the nodes in the ROS system, to run it:
	roslaunch osr_bringup osr.launch"

Running this gives error message (regardless of which directory we run it from):
     	[osr.launch] is neither a launch file in package [ors_bringup] nor is [osr_bringup] a launch file name
so we suspect that some ROS build needs to be done, but aren't sure what.

JHP

ireasor, 2019-06-10:
Hi @JHPHELAN.

I haven't tried to build the ROS codebase yet, so I can't help you here.  The only thing I've done with this is grab the code for the Arduino and load it.

JHPHELAN, 2019-06-11:
@ireasor 
Can you enumerate the steps you took to grab and load the Arduino code?
JHP

ireasor, 2019-06-11:
@JHPHELAN We discussed that in issue #24 - It looks like @ericjunkins merged my pull request to the branch, so you may want to pull down the latest version and try again?  We should probably limit any further discussion about the Arduino code to that thread, though.

JHPHELAN, 2019-06-16:
@ireasor @ericjunkins @vssystemluba 
[The current directions fail...]
     On your Raspberry Pi run the following commands in a terminal
     mkdir -p ~/osr_ws/src
     cd ~/osr_ws/src
     git clone https://github.com/nasa-jpl/osr-rover-code/tree/osr-ROS/ROS
[...with the the error:]
     Cloning into 'ROS'...
     fatal: repository 'https://github.com/nasa-jpl/osr-rover-code/tree/osr-ROS/ROS/' not found

[ireasor's directions work:]
$ mkdir -p ~/osr_ws/src
$ cd ~/osr_ws/src
$ git clone https://github.com/nasa-jpl/osr-rover-code.git
[the missing step here is to drill down from the home directory into the osr-rover-code directory:]
$ cd ~/osr_ws/src/osr-rover-code
[then switch branches with:]
$ git checkout osr-ROS
$ git pull
[this then shows successful listing:]
~/osr_ws/src/osr-rover-code$ ls
Arduino  img  LICENSE.txt  README.md  ROS
[which when drilling down to ROS gives:]
~/osr_ws/src/osr-rover-code/ROS$ ls
led_screen  osr  osr_bringup  osr_msgs

[In later directions:]
Running
There is a roslaunch file that will start all of the nodes in the ROS system, to run it:
roslaunch osr_bringup osr.launch
[gives the error:]
$ roslaunch osr_bringup osr.launch
[osr.launch] is neither a launch file in package [osr_bringup] nor is [osr_bringup] a launch file name
The traceback for the exception was written to the log file

[Here my ignorance of ROS comes out.  I SUSPECT that at this point we need to do some sort of build or make or source command but I'm still learning.  Any guidance here??]

Jim Phelan
Mars rover project lead
USAi Labs (formerly Houston Robotics)

ericjunkins, 2019-06-16:
Yeah there is a little bit to Setup the ROS system etc. We'll try and get these done before the builders call this coming week

mikcox, 2019-08-22:
Closing this due to inactivity; this issue was also addressed in the master open-source-rover branch.

closed 2019-08-22 by mikcox