nasa-jpl/osr-rover-code #24: osr-ROS - Unable to load Arduino code

ireasor, 2019-05-18:
I'm trying to follow the steps in the osr-ROS README to load the code onto my Arduino.  After my workaround for issue #23, I got the code downloaded and I got the sample project loaded into the Arduino IDE.  When I try to upload this to my Arduino, though, I get the following error:

Osr_Screen:10:28: error: RGBmatrixPanel.h: No such file or directory
compilation terminated.
exit status 1
RGBmatrixPanel.h: No such file or directory

Looking through the source code, I can confirm that there is no RGBMatrixPanel.h file present.

ericjunkins, 2019-05-20:
@ireasor Ah, you are correct  I need to add that step as well. That library is directly from the adafruit tutorial for the led matrix. It can be found at:

https://learn.adafruit.com/32x16-32x32-rgb-led-matrix/test-example-code

JHPHELAN, 2019-05-25:
@ireasor How did you get the Arduino code loaded?  A step-by-step would help the rest of us as well as the JPL team's documentation!
JHP

ireasor, 2019-05-26:
@JHPHELAN See PR #25 for updated steps.

JHPHELAN, 2019-05-26:
@ireasor I'm confused because I don't know Github all that well.  PR#25 says it's been updated, but I can't find the updated instructions.  :-(  The electrical build and software steps in the new electrical-pcb-rework branch have no mention of Arduino.  The last update on anything I can see was 18d ago.
Can you point me in the right direction?  Thanks!!
JHP

ireasor, 2019-05-29:
@JHPHELAN if you take a look at https://github.com/ireasor/osr-rover-code/tree/osr-ROS, you can see my version of the repository that I submitted the pull request for.  If you take a look at https://github.com/nasa-jpl/osr-rover-code/pull/25/files, you can see the changes that I have made.

JHPHELAN, 2019-05-29:
@ireasor Ah, I'm beginning to see how this works!  Thanks!!
JHP

JHPHELAN, 2019-05-31:
@ireasor When I follow the directions in the update above, when I try to upload the sketch I get the following error message:
Arduino: 1.8.9 (Windows 10), Board: "Arduino/Genuino Uno"

In file included from C:\Users\Me\Documents\Arduino\libraries\Arduino\examples\Osr_Screen\Osr_Screen.ino:12:0:

C:\Users\Me\Documents\Arduino\libraries\Arduino/OsrScreen.h:4:55: fatal error: ../RGB-matrix-Panel-master/RGBmatrixPanel.h: No such file or directory

compilation terminated.

exit status 1
Error compiling for board Arduino/Genuino Uno.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

JHP (world expert in destructive testing of software)

ireasor, 2019-05-31:
If you take a look at the changes I made, you'll see that I had to modify OsrScreen.h and OsrScreen.cpp to remove "-master" from the path.  Hopefully it works the same for you on Windows.

JHPHELAN, 2019-06-01:
@ireasor Ian - REALLY appreciate your guidance as I'm a total Arduino newbie!  Ok, I took out the -master part and reloaded the zip library.  Now, even though the RGBmatrixPanel.h is in the very next tab on the Arduino IDE, I get the following error message near the bottom (Maybe I should turn off 'verbose'?): 
Thanks again,
JHP

Arduino: 1.8.9 (Windows 10), Board: "Arduino/Genuino Uno"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Me\Documents\Arduino\libraries -fqbn=arduino:avr:uno -vid-pid=0X2341_0X0043 -ide-version=10809 -build-path C:\Users\Me\AppData\Local\Temp\arduino_build_438077 -warnings=default -build-cache C:\Users\Me\AppData\Local\Temp\arduino_cache_833766 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\Me\Documents\Arduino\libraries\Arduino\examples\Osr_Screen\Osr_Screen.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Me\Documents\Arduino\libraries -fqbn=arduino:avr:uno -vid-pid=0X2341_0X0043 -ide-version=10809 -build-path C:\Users\Me\AppData\Local\Temp\arduino_build_438077 -warnings=default -build-cache C:\Users\Me\AppData\Local\Temp\arduino_cache_833766 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\Me\Documents\Arduino\libraries\Arduino\examples\Osr_Screen\Osr_Screen.ino
Using board 'uno' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Detecting libraries used...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\standard" "C:\\Users\\Me\\AppData\\Local\\Temp\\arduino_build_438077\\sketch\\Osr_Screen.ino.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\standard" "-IC:\\Users\\Me\\Documents\\Arduino\\libraries\\Arduino" "C:\\Users\\Me\\AppData\\Local\\Temp\\arduino_build_438077\\sketch\\Osr_Screen.ino.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\standard" "-IC:\\Users\\Me\\Documents\\Arduino\\libraries\\Arduino" "-IC:\\Users\\Me\\Documents\\Arduino\\libraries\\Adafruit_GFX_Library" "C:\\Users\\Me\\AppData\\Local\\Temp\\arduino_build_438077\\sketch\\Osr_Screen.ino.cpp" -o nul
In file included from C:\Users\Me\Documents\Arduino\libraries\Arduino\examples\Osr_Screen\Osr_Screen.ino:12:0:

C:\Users\Me\Documents\Arduino\libraries\Arduino/OsrScreen.h:4:48: fatal error: ../RGB-matrix-Panel/RGBmatrixPanel.h: No such file or directory

compilation terminated.

Using library Arduino in folder: C:\Users\Me\Documents\Arduino\libraries\Arduino (legacy)
Using library Adafruit_GFX_Library at version 1.5.2 in folder: C:\Users\Me\Documents\Arduino\libraries\Adafruit_GFX_Library 
exit status 1
Error compiling for board Arduino/Genuino Uno.

JHPHELAN, 2019-06-02:
@ireasor 
One step closer -
Looked again at the error message:
../RGB-matrix-Panel/RGBmatrixPanel.h: No such file or directory
Looking in the folder - C:\Users\Me\Documents\Arduino\libraries\Arduino there IS a folder RGB_atrix_Panel and it DOES contain RGBmatrixPanel.h.  

Oh!  Wait!  It’s looking for RGB-matrix-Panel but it’s finding RGB_matrix_Panel.  A case of ‘-‘ vs ‘_’!
Not sure on which side of the conflict the change needs to be made in the documentation. (along with the -master change)

Renaming ‘_’ to ‘-‘ and trying again....  This time I get (being less verbose this time):
    Osr_Screen:10:28: error: RGBmatrixPanel.h: No such file or directory
but there IS RGBmatrixPanel.h in the
     C:\Users\Me\Documents\Arduino\libraries\Arduino\RGB-matrix-Panel
folder!

Stumped again,
JHP

ericjunkins, 2019-06-13:
@vssystemluba Can you help assist with this issue, I'm swamped with stuff

JHPHELAN, 2019-06-15:
Started from scratch by re-installing Arduino IDE & re-downloading .zip of github and re-uploading to Arduino.  Not sure what the display is supposed to look like, but at least It downloaded without error and displayed this (open on your own PC and rotate to see clearly):
;-))
JHP

[image]

ericjunkins, 2019-06-20:
It appears that this is solved now right? I'm going to close this

closed 2019-06-20 by ericjunkins