nasa-jpl/osr-rover-code #33: roslaunch node boot sequence

ericjunkins, 2019-08-18:
The rosnodes need a boot sequence forced to make sure that the motor_controller node completes its' initialization before other nodes start

Achllle, 2020-02-15:
ROS is distributed and thus you cannot enforce which nodes run when. The way to deal with this is to wait for certain services or topics or so to come up before you continue.

ericjunkins, 2020-02-17:
It's not strictly enforced no, this was to make sure that the motor_controller was fully initialized before any parameters were passed to the other nodes, which can be monitored by flags in the system. What it should do is the motor controller node, and wait until all it is pulled/pushed all the necessary operating parameters to it, and then give the go-ahead flag to the rest of the nodes the begin their initialization.

Achllle, 2020-06-16:
No longer relevant

closed 2020-06-16 by Achllle