nasa-jpl/osr-rover-code #34: Serial Port device name

ericjunkins, 2019-08-18:
the motor_controller node takes parameter of a serial device to use on the RPi to communicate to the motor controllers. This should be /dev/ttyS0 or /dev/serial0. Also might think about having it try /dev/ttyAMA0 or /dev/serial1 if those fail.

Achllle, 2020-02-15:
Using udev rules would also deal with this quite nicely

ericjunkins, 2020-02-17:
Either a udev rule or a smiple try/catch to retry the other device port would be fine. we know it can only be one of two.

Achllle, 2020-02-17:
Happy to take this one

Achllle, 2020-03-07:
Looking into this a little bit more, I think there are two options here:

* Let the user decide what device to talk to, through a rosparam, see #42. Set the default to `/dev/serial0`, as only in rare cases should `/dev/serial1` be used. The symlinks should always be used as opposed to the original device files `S0` and `AMA0`.
* remove the rosparam, try/catch using both device names. This is easier, but obscures things and more code is required (`rc.Open` will work on both devices, so you would have to try to read data on both devices and see which one responds. If the wrong one is tried first, this would add a several second delay to the launch each time)

mikcox, 2020-03-28:
This issue is resolved as of the merge of #42.

closed 2020-03-28 by mikcox