nasa-jpl/osr-rover-code #35: Software switch of left-right of the robot.

ericjunkins, 2019-08-18:
It appears that there is a discrepancy between the hardware docs assigning left/right of the robot and what the code currently has. This can be fixed by inverting the list directions on all the functions that calculate the speeds and the corner angles. 

_As a note about this. The hardware I have built on my robot is such that the software is correct for it, it must have gotten put into the hardware docs incorrect somewhere. As I don't want to redo all the hardware on my robot my proposition is to add a variable managed to do left-right swap for the robot._

Achllle, 2020-03-11:
Wish I'd seen this and the tapatalk post earlier! I fixed this independently. I'll open a PR with a fix. I'm happy to include a variable but it seems like you'd be the only one setting it differently and since the extra variable means extra documentation and a source for confusion, I would say the recommended approach would be for you to have a separate commit with a revert on your fork and have the default be the one that follows the existing build docs, especially since several people have built it the 'new way'.

Your call. Let me know, but this needs to get fixed rather sooner than later.

Achllle, 2020-06-16:
closing because no longer relevant

closed 2020-06-16 by Achllle