nasa-jpl/osr-rover-code #46: Don't include roboclaw.py ourselves

Achllle, 2020-03-07:
Instead of duplicating the source for the roboclaw code (`roboclaw.py`), we should clone the [official repo](https://github.com/basicmicro/roboclaw_python_library), perhaps fork it to turn it into a catkin or pip package. 

This ensure we can get updates properly, keep our codebase smaller and cleaner, and report issues related to roboclaw code in their repo where we will find official support.

Achllle, 2020-03-17:
Or, since the official repo isn't the greatest of quality and has zero documentation or support, it might make sense to turn to user repos:

* https://github.com/csvance/roboclaw c++, intended support for this repo
* https://github.com/sonyccd/roboclaw_ros not maintained, seems for diff drives only
* https://github.com/urill/pyroboclaw less attractive

Achllle, 2020-04-01:
I'm working on this at https://github.com/Achllle/libroboclaw and https://github.com/Achllle/roboclaw-hardware-interface

ericjunkins, 2020-04-01:
roger 👍

JHPHELAN, 2023-07-04:
strange roboclaw_wrapper error messages in foxy-devel
Started to get odd roboclaw_wrapper error messages on launch.
Completely reinstalled OSR per foxy-devel branch but errors persist.
roboclawtest.py works fine on all 5 RoboClaws.  
BasicMicro Motion Studio checks all 5 ok.

'''
[roboclaw_wrapper-1] [INFO] [1688488053.574796233] [roboclaw_wrapper]: Initializing motor controllers
[roboclaw_wrapper-1] [INFO] [1688488053.766564605] [roboclaw_wrapper]: Sucessfully connected to RoboClaw motor controllers
[rover-2] [INFO] [1688488054.755269628] [rover]: Initializing Rover
[roboclaw_wrapper-1] [ERROR] [1688488058.242325643] [roboclaw_wrapper]: Motor controller '128' reported error code 2147844096
[roboclaw_wrapper-1] [ERROR] [1688488058.247449208] [roboclaw_wrapper]: Motor controller '129' reported error code 2147844096
[roboclaw_wrapper-1] [ERROR] [1688488058.252542513] [roboclaw_wrapper]: Motor controller '130' reported error code 2147844096
[roboclaw_wrapper-1] [ERROR] [1688488058.258753801] [roboclaw_wrapper]: Motor controller '131' reported error code 3221520384
[roboclaw_wrapper-1] [ERROR] [1688488058.265225627] [roboclaw_wrapper]: Motor controller '132' reported error code 2147844096
[roboclaw_wrapper-1] Traceback (most recent call last):
[roboclaw_wrapper-1]   File "/home/ubuntu/osr_ws/install/osr_control/lib/osr_control/roboclaw_wrapper", line 11, in <module>
[roboclaw_wrapper-1]     load_entry_point('osr-control', 'console_scripts', 'roboclaw_wrapper')()
[roboclaw_wrapper-1]   File "/home/ubuntu/osr_ws/build/osr_control/osr_control/roboclaw_wrapper.py", line 546, in main
[roboclaw_wrapper-1]     rclpy.spin(wrapper)
[roboclaw_wrapper-1]   File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/__init__.py", line 191, in spin
[roboclaw_wrapper-1]     executor.spin_once()
[roboclaw_wrapper-1]   File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/executors.py", line 719, in spin_once
[roboclaw_wrapper-1]     raise handler.exception()
[roboclaw_wrapper-1]   File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/task.py", line 239, in __call__
[roboclaw_wrapper-1]     self._handler.send(None)
[roboclaw_wrapper-1]   File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/executors.py", line 429, in handler
[roboclaw_wrapper-1]     await call_coroutine(entity, arg)
[roboclaw_wrapper-1]   File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/executors.py", line 343, in _execute_timer
[roboclaw_wrapper-1]     await await_or_execute(tmr.callback)
[roboclaw_wrapper-1]   File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/executors.py", line 118, in await_or_execute
[roboclaw_wrapper-1]     return callback(*args)
[roboclaw_wrapper-1]   File "/home/ubuntu/osr_ws/build/osr_control/osr_control/roboclaw_wrapper.py", line 218, in slow_update
[roboclaw_wrapper-1]     self.status.error_status = self.read_errors()
[roboclaw_wrapper-1]   File "/home/ubuntu/osr_ws/install/osr_interfaces/lib/python3.8/site-packages/osr_interfaces/msg/_status.py", line 178, in error_status
[roboclaw_wrapper-1]     assert \
[roboclaw_wrapper-1] AssertionError: The 'error_status' field must be a set or sequence with length 5 and each value of type 'int' and each integer in [-2147483648, 2147483647]
[ERROR] [roboclaw_wrapper-1]: process has died [pid 2612, exit code 1, cmd '/home/ubuntu/osr_ws/install/osr_control/lib/osr_control/roboclaw_wrapper --ros-args -r __node:=roboclaw_wrapper --params-file /home/ubuntu/osr_ws/install/osr_bringup/share/osr_bringup/config/roboclaw_params.yaml'].
'''
Stops here and fails to respond to X-box controller though paired.