nasa-jpl/osr-rover-code #55: osr_msgs: list --> fields

Achllle, 2020-03-07:
Currently the messages contain an ordered list for each motor which is hard to interpret and error prone. Either the messages should be converted to contain specific fields, e.g.:

```
drive:
  int32 left_front
  int32 left_middle
  int32 left_back
  int32 right_front
  int32 right_middle
  int32 right_back
steer:
  int32 left_front
  int32 left_back
  int32 right_front
  int32 right_back
```

OR the order should be better documented. I believe currently the order is 131M1, 131M2, 132M1, 132M2. Perhaps we want to change this to be clockwise starting front right or similar.

Achllle, 2020-03-07:
We can also use the commonly used `sensor_msgs/JointState` message which includes naming, position, velocity, effort. And works with JointStatePublisher as well as [RobotStatePublisher](http://wiki.ros.org/robot_state_publisher) nodes which we'll need to do anyways if we want to set up ROS tf2 and RViz and the like easily

Achllle, 2020-03-07:
Similarly, the rosparams should also not be lists which are ambiguous.

ericjunkins, 2020-03-25:
I agree, each of these should have individual field names, instead of lists. It was convenient to do things as lists initially, however is confusing and ambiguous as to what is being accomplished at each step.

PR #71 (2020-03-31): Overhaul code towards a more ROSic system
## Description

Quite large overhaul of the code. Many improvements, tracked through the following issues:

* close #65
* close #64 
* close #60
* close #59 
* close #55 
* close #53 
* close #41 
* close #36 
* greatly simplify math on corner and drive motor velocity/position calculation
* improve code quality

## Dependencies

- [x] #52 
- [x] #57
commit 2c986b67 (2020-03-31): Overhaul of roboclaw_driver.py, rover.py

Addresses:
* close #64
* close #60
* partial work towards #58
* close #55
closed 2020-03-31 by ericjunkins