nasa-jpl/osr-rover-code #59: Consider switching to SI units

Achllle, 2020-03-08:
To be consistent with the robotics and ROS community, it would make sense to switch to SI units for all calculations. See ROS' [REP103](https://www.ros.org/reps/rep-0103.html). Instead of degrees and degrees per second, we would use radian and rad/s as well as meter and m/s iso inches.

Happy to do this if there is consensus on this.

ericjunkins, 2020-03-25:
This makes sense to do yes. Initially units of inches were chosen as parts are all english-unit parts, and thus was easier-ish in that regard. But when the transition to ROS happened SI unit conversion should have happened as well.

Achllle, 2020-03-25:
Opening a PR for this once #48 and some other are in.

PR #71 (2020-03-31): Overhaul code towards a more ROSic system
## Description

Quite large overhaul of the code. Many improvements, tracked through the following issues:

* close #65
* close #64 
* close #60
* close #59 
* close #55 
* close #53 
* close #41 
* close #36 
* greatly simplify math on corner and drive motor velocity/position calculation
* improve code quality

## Dependencies

- [x] #52 
- [x] #57
closed 2020-03-31 by ericjunkins