nasa-jpl/osr-rover-code #62: Use diagnostics_msgs/DiagnosticStatus.msg iso custom Status message

Achllle, 2020-03-20:
Currently the message `osr_msgs/Status` contains:
```
int64     battery
int64[5]  error_status
int64[5]  temp
int64[10] current
```

I propose:
* `battery` should have its own topic, with message `sensor_msgs/BatteryState`
* `error_status` should be reported in `/diagnostics` using `diagnostics_msgs/DiagnosticArray` which would allow the use of [diagnostics tools](http://wiki.ros.org/diagnostics)
* `temperature` should have its own topic, with message type `sensor_msgs/Temperature`
* `current` should be reported in the same `sensor_msgs/JointState` message used to report feedback on the wheel positions and velocities under 'effort'.

ericjunkins, 2020-03-25:
My one thought about this change is that now if you want to look at the overall health of the robot you have to look in many different places. I would like to maintain the ability to query one single location to get debug information about the robot, however I think there is also benefit to having the topics you proposed as well for other tools. I think it might make sense to just publish to both. In addition to this i agree we should clean up the message types used for these.

Achllle, 2020-03-25:
Aggregation is typically solved using `diagnostics_aggregator` and [robot_monitor](http://wiki.ros.org/diagnostics/Tutorials/Using%20the%20Robot%20Monitor) like systems. The nice thing about it is that people can choose what tool they want to use for that. For debugging, I typically will set the log_level of whatever node I'm debugging to `DEBUG` and I'll get all the relevant information that way. I can also see the value of having both, but that's more stuff to maintain, especially if things change.

itstalmeez, 2023-10-22:
Like definitely its my opinion you are not supposed to go with this so, separating 'battery' into its own topic with sensor_msgs/BatteryState is a great idea. This will provide more detailed and standardized information about the battery status.

Moving 'error_status' to /diagnostics using diagnostics_msgs/DiagnosticArray is a wise decision. It promotes a cleaner separation of concerns, enabling the use of diagnostic tools for better error management.

Creating a separate topic for 'temperature' using sensor_msgs/Temperature will enhance the clarity of data and its intended use.

Including 'current' in the sensor_msgs/JointState message for wheel positions and velocities is efficient, as it groups related data together.

These changes will not only enhance the modularity of your project but also foster interoperability with other ROS packages.

Achllle, 2023-10-24:
@itstalmeez you're welcome to create a PR!