nasa-jpl/osr-rover-code #64: tick2deg assumes 90 deg range for corner motors

Achllle, 2020-03-21:
Although easy to calculate and independent of encoder model,

* this method can only be used for corner encoders as it assumes a range of 90 degrees
* the range isn't exactly 90 degrees, due to mechanical inaccuracies and soft limits.

My proposal is to convert ticks based on the constant ration tick/radian or tick/degree (related: #59) and have a setting for this ratio for those who have different encoders/motors.

ericjunkins, 2020-03-25:
Sorry I am unsure of your exact proposal here. I think we should look into changing this function however. Your point on assuming it is 90 degrees is valid. I think this degree window should be taken as a ros-param, as I know some people already have, or want to, change the total corner turning range. Can you explain further what your idea of what you want to change is?

Achllle, 2020-03-25:
Sorry I didn't explain well. 
Based on encoder/motor specs you can figure out what the conversion from ticks to position (degrees or radian) is and vice versa as the motors have a fixed gear ratio and encoders have a certain number of ticks per revolution. 
The reason you want to do this is because it's more accurate than the current method (assuming max_enc - min_enc corresponds to 45deg - (-45deg)). Currently, the range for all corner motors is assumed to be -45 -> 45 degrees. In actuality, due to tolerances in assembly and 3D printed parts, due to inaccurate determination of the encoder min and max values, and due to the recommended soft limit (add or subtract 15 encoder ticks from the range), this range isn't exact. 
Rather than exposing the number of ticks as the API for commanding and reading motor positions and velocities, I also converted ticks/qpps to position/velocities at the lower level (`roboclaw_wrapper.py`) so that commanding motors is uniform and independent of the motor, and we can abstract roboclaw internals away and simplify code.

I've addressed this in [my iss45 branch](https://github.com/Achllle/osr-rover-code/tree/iss45). I exposed the motor settings in a `yaml` file which is loaded into rosparam.

ericjunkins, 2020-03-28:
Just to have a track of this, after more discussion with @Achllle we settled on this being a good idea. To outline exactly what this is going to do:

- Use the encoder max/min values that are set during the calibration phase to find the  0 radian mark
- From there position commands are calculated by calculating the encoder ticks for a given radian target, by using the gear ratio and encoder ticks/revolution

Feel free to correct if I got any of that wrong Achllle

PR #71 (2020-03-31): Overhaul code towards a more ROSic system
## Description

Quite large overhaul of the code. Many improvements, tracked through the following issues:

* close #65
* close #64 
* close #60
* close #59 
* close #55 
* close #53 
* close #41 
* close #36 
* greatly simplify math on corner and drive motor velocity/position calculation
* improve code quality

## Dependencies

- [x] #52 
- [x] #57
commit 2c986b67 (2020-03-31): Overhaul of roboclaw_driver.py, rover.py

Addresses:
* close #64
* close #60
* partial work towards #58
* close #55
closed 2020-03-31 by ericjunkins