nasa-jpl/osr-rover-code #66: Use velocity control on corner motors

Achllle, 2020-03-25:
Instead of using position control, use a velocity controller (or even torque) on the corner steering motors. This would allow higher control loop rates to ensure that all wheels are synced up while turning.
Perhaps also the time to switch to ros_control

ericjunkins, 2020-03-25:
When you are saying use a velocity control on the corner motors, can you clarify at what level you are meaning to implement this controller on? Are you meaning use a velocity controller on the motor controller? On the ROS side and wrap everything underneath it?

"All synced up" for the turning does not necessarily mean that they all are moving the same speed however. It means that they move through their delta between start and target at the same linear division of the range. Take the example where while you are driving you want to switch from hard left to hard right turn. For a perfect case the sync of the wheels makes this transition a continuous movement of the focal point, which would have to be done by looking at the delta of all corners, and setting the turning rate proportional to the amount of delta on each.

Achllle, 2020-04-02:
I'm overloading a few things on top of each other making my post a bit misleading.

When I write velocity control I mean having velocity as the control input, not as the setpoint. What I want to address in this issue is that the corner motors don't turn at the same rate. Just sending setpoints to each of the controllers means that they won't sync up. Ros control is helpful for something like this, as you can send trajectories for all corner motors using the joint trajectory controller and it has all the machinery built in for switching to new goals smoothly.

ericjunkins, 2020-04-02:
Oh gotcha,  that would be a nice modification yes!

PR #80 (2020-10-12): use current radius iso commanded radius to calculate drive velocities
## Description

Taking feedback on a way to calculate the current radius given encoder readings. I implemented a simple way of doing this using the median value of the calculated radii according to each motor position. There's probably other good ways of doing this. 
close #69 , close #66 
The idea behind calling the method `forward_kinematics` is that I intend to calculate the current twist at each time step so that we can keep track of odometry easily in perhaps a follow-up PR.

To get this to work correctly, I also had to [... cut]
closed 2020-10-12 by Achllle