nasa-jpl/osr-rover-code #68: Create robot_description package

Achllle, 2020-03-26:
The robot_description package contains the URDF/XACRO files used to represent the robot which is useful for visualization in RViz, running the robot in simulation with Gazebo, and sensor placement.

ericjunkins, 2020-03-28:
I can look into digging what I made for this a longgg time ago up. We do have this, somewhat frameworked out already

Achllle, 2020-06-16:
@ericjunkins can you dig this up before it eternally gets lost?

ericjunkins, 2020-06-17:
I'll try and get to this on the upcoming friday

Achllle, 2020-08-12:
Got a first version using the sw_exporter package. The most important thing is having the configuration saved in SolidWorks so we can iterate on it and keep exporting as the model updates.
The tool doesn't work very well and so I need to iterate on it a few times to get all the axes right.

soldierofhell, 2021-06-05:
Hi @Achllle, can you upload anything you have so far that can help in robot_description creation? Regards,

Achllle, 2021-06-15:
All work lives in master in the form of SW URDF exporter configuration files. You'll need SolidWorks to be able to use them.

Achllle, 2025-01-02:
closed by #170 and #193

closed 2025-01-02 by Achllle