nasa-jpl/osr-rover-code #70: Disable autorepeat of joy messages

Achllle, 2020-03-26:
For some reason the joystick continuously publishes the last message it received to the rover. As a result, when you're driving forward at full speed and the joystick disconnects, the rover will keep going in whatever the last direction was.

The fix is as simple as setting the `autorepeat_rate` in osr.launch for the joy node back to 0.0, or simply deleting that line.

ericjunkins, 2020-03-28:
I remember there being some case where it seemed like a good idea to have it, but am trying to remember the exact case. I think it was something similar, but to the effect of if the controller lost connection while you were moving the joy stick, which can happen if you are a bit far away from it, joy wouldn't fire a new command. The case here that happens is:

Someone is telling the robot to drive forward, it starts getting to the limit of the controller, they tell it to stop going forward but that packet is dropped. We wanted it to repeat the values it's at next to have it stop driving or drive backwards etc.

Achllle, 2020-03-29:
What you want instead is that the controller stops the wheels whenever it doesn't get a command (a heartbeat) for a certain period of time. The joystick node only has to send commands when there are commands to be sent. When a command doesn't make it, no big deal, the Rover stops moving. But at least it's not driving full speed into an obstacle

ericjunkins, 2020-03-30:
But that's only true if you are sending new updated signals every heartbeat timeperiod right? For instance, if you were to command the robot to move forward at full speed and just left it there, without moving the joystick it would timeout on the heartbeat because it didn't get a new command correct?

Achllle, 2020-03-30:
I was (partially*) incorrectly assuming that joy node didn't recognize that the controller has been disconnected and keeps publishing the last command instead of just publishing defaults. Instead I was running into the bug that I described where a network disconnection magically freezes up the driver (tbd).

(*) [joy node wontfix for different behavior of autorepeat for different controllers](https://github.com/ros-drivers/joystick_drivers/issues/92). 

We have to be careful with this setting. A safer approach would be to force autorepeat yourselves by repeatedly pushing a random button or slightly moving the joystick while driving. 
In any case, the roboclaw_wrapper should probably still have an auto-turnoff when no commands are received for a certain time. A joystick won't always be the only node sending commands (e.g. autonomous mode or different publisher), could crash or shut down by accident and so the rover could still crash in that case. That's a somewhat separate issue though. Closing this for now.

closed 2020-03-30 by Achllle

ericjunkins, 2020-03-30:
I agree on your point here. Can think about that moving forward 👍