nasa-jpl/osr-rover-code #109: support for ROS2

Achllle, 2020-08-03:
A meta-issue for supporting ROS2. ROS2 is slowly becoming the new default and folks will be trying to switch over over the next months/years. This is a good opportunity to become familiar with the new standard and attract more developers.

Feel free to chime in with thoughts.

ericjunkins, 2020-08-03:
Are you thinking we should support both ROS and ROS2?

Achllle, 2020-08-03:
I'm thinking we should have a branch for ROS2 with ROS1 our default. Any changes that apply to both can be backported. And then if ROS2 becomes mature and we fully support it, we can decide to switch over. But we have to put in the effort first, which I'm happy to do

Achllle, 2020-08-10:
Initial work on branch [Achllle/osr-rover-code/iss109_ros2fox](https://github.com/Achllle/osr-rover-code/tree/iss109_ros2foxy)

Achllle, 2021-03-26:
Closing since there is now proper ROS2 support - decision on making that the main branch will follow later

PR #126 (2021-03-26): Convert code base to ROS2 Foxy
close #109 

Note the target branch. At a later date we can make foxy-devel the default branch and do backports to `melodic-devel` and `noetic-devel`.

See commits and the ROS2 tutorials to make sense of the changes.

## TODO

- [x] Change installation instructions to Ubuntu20.04
- [x] Check systemd scripts
- [ ] ~~Led screen version~~
- [x] convert joy launch
closed 2021-03-26 by Achllle