nasa-jpl/osr-rover-code #110: Add speed multiplier to ros param_mod

ericjunkins, 2020-08-12:
We should add a field to be read in from the physical_properties_mod.yaml to be the speed multiplier for the top speed desired by the robot for each user. Because a lot of us are using different motors with different RPMs the multiplier should be different for easy user.

Achllle, 2020-09-03:
Perhaps this is just a documentation issue? You can already do this by setting the following values appropriately in `launch/bringup.launch`:

```
            <param name="scale_linear" value="0.24"/>  <!-- scale to apply to drive speed, in m/s: drive_motor_rpm * 2pi / 60 * wheel radius -->
            <param name="scale_angular" value="0.53"/>  <!-- scale to apply to angular speed, in rad/s: scale_linear / min_radius -->
            <param name="scale_linear_turbo" value="0.5"/>  <!-- scale to apply to linear speed, in m/s -->
```

Achllle, 2020-11-15:
@ericjunkins

Achllle, 2021-07-16:
closing this because I think you can do this today and it's documented. If you think that's incorrect, pls reopen

closed 2021-07-16 by Achllle