nasa-jpl/osr-rover-code #111: Joystick node in osr.launch does not allow commands to LED screen

dcschooley, 2020-08-13:
It looks like the "joy2twist" node defined in osr.launch is intended to replace the joystick.py module. The new definition of the joystick does not allow commands to be sent to the LED screen,

ericjunkins, 2020-08-13:
Ah, hadn't thought of that. We should be able to still subscribe to the joy node and just capture button presses to send to the rostopic for led

Achllle, 2020-09-03:
My bad, I don't have the LED screen so didn't know this was a thing!
I can attempt to write such a node, but it's impossible for me to test, so I'll need some help here.

JHPHELAN, 2020-10-14:
I didn't know using buttons on the Xbox to control the LED was even a thing!  I didn't see anything about that in the docs.
In fact, I don't really believe there are any instructions in the docs about how to actually use the Xbox to drive the rover.  I've picked up about using the left trigger as a dead man switch but not really sure where.  I have the LED screen mounted on the front of my rover instead of the head.  Would be glad to do some testing once we get the Xbox to communicate.

Achllle, 2020-10-14:
Instructions are [here](https://github.com/nasa-jpl/osr-rover-code/blob/master/setup/rover_bringup.md#1-manual-rover-bringup). I assume you mean 'left button (LB)'? The trigger is the one below it. It could also be that your xbox controller has a different button mapping, which you can control in the `osr.launch` file.

Have you figured out what RB does? :wink:

dcschooley, 2020-10-14:
Yeah, I think the documentation could be improved regarding the buttons. It took me a while to find the correct button to hold down to make the motors go. I think I had to dig through the code and then figure out which button number corresponded to which button. That is a good exercise, but it's a bummer when you are all excited to have things basically finished and you think your rover should be working but it isn't...

JHPHELAN, 2020-10-14:
Whoa!!  Just tried roslaunch and the Xbox again.  This time trying the Left Button instead of what I now understand was Left Trigger.  A diagram of the Xbox controller would be helpful as would an explanation of which joysticks do what.  I mean, it's easy enough to fumble around and figure it out, but....  Sudden case of Parkinson's with all the steering motors!! (I can see some more calibration in the future!)  I am using the REV HEX motors for drive motors instead of the (2nd) standard ones.   Zooooom!!!  I think the Right Button was a secret speed boost in the previous version.  Now the motors run so fast I don't see a difference.

Did the buttons affect the LED before?  What did they do?  Or was that just custom stuff?

SallyTYT, 2022-09-28:
> My bad, I don't have the LED screen so didn't know this was a thing! I can attempt to write such a node, but it's impossible for me to test, so I'll need some help here.

Hi @Achllle , I'm facing this issue now, could you work with me to bring up the LED again?

closed 2023-09-25 by Achllle