nasa-jpl/osr-rover-code #131: Add joystick permission setting instructions for ROS2

Achllle, 2021-03-27:
## input group option

add $USER to input

# modify dev permissions directly

1. `ls -l /dev/input/*`
2. `chmod a+rw /dev/input/event0` or whatever corresponds to the event joystick
3. make a udev rule like `KERNEL=="event[0-9]*", ATTRS{idVendor}=="045e", MODE:="0666", SYMLINK+="xbox"`

apollokit, 2021-03-27:
Or maybe a better option: just add the user to the `input` group: `sudo adduser ubuntu input` (and then log out/in again)

The reason why is we need access to `/dev/input/event*`, in addition to the js* devices. The event* are in the input group by default.

idgaron, 2025-05-29:
Is there a reason this was never added to the documentation? Seems like a reasonable change to add the user to the input group, and it's what I did for my rover

Achllle, 2025-05-29:
Combination of not knowing/remembering that this is an issue for other folks and all of us being volunteers with some elements of being lazy engineers? :) PR most welcome.

idgaron, 2025-05-29:
lol no worries - I have done the same thing countless times. just trying to clean up the documentation after having built one fresh myself to give other people an easier time

PR #216 (2025-05-29): Add user to Input Group to enable joystick without changing permissions every time
The Joystick is not read by the user by default since the input file does not have read permissions for other users. Adding the user to the Input group gives them read access to the joystick inputs.

closes #131
closed 2025-05-29 by Achllle