nasa-jpl/osr-rover-code #134: Seeing ZeroDivisionError when odometry is enabled

veeresht, 2021-05-02:
[image]

Seeing this ZeroDivisionError when the rover is brought up with `enable_odometry:=true` option, which indicates that `self.curr_turning_radius` is 0 or close to 0.

Achllle, 2021-05-05:
Thanks for the find. I am unable to reproduce this but I can see the edge case where this would happen. I will submit a simple patch with a try-except wrapper as a fix ASAP.

Achllle, 2021-05-08:
This happens when the angle of the wheels is +/- 55.4 degrees which is an invalid value since the max angle is around 45 degrees. You should check your wheel calibration if you're building an unmodified rover.

apollokit, 2021-05-24:
i forget, @Achllle you closed this out with a pr right?

Achllle, 2021-05-24:
Correct

commit c8a7ea60 (2021-05-24): Merge pull request #136 from Achllle/iss134_odomzerodiv

Fix #134 for ROS1, add warning
commit d8d32f13 (2021-05-24): Fix #134 for ROS1, add warning
closed 2021-05-24 by Achllle