nasa-jpl/osr-rover-code #145: change editing of config.txt to usercfg.txt

JHPHELAN, 2021-08-21:
In section: 
5.5 Disable bluetooth in config.txt boot config file
Execute the following commands
	cd /boot/firmware
	sudo cp config.txt config.txt.bak
	sudo nano config.txt
And then add the new line dtoverlay=disable-bt immediately after the existing line cmdline=cmdline.txt towards the bottom of the file.
PROBLEM – there is no cmdline=cmdline.txt line in the file.
This has been changed to usercfg.txt
config.txt has the instruction:
# Place "config.txt" changes (dtparam, dtoverlay, disable_overscan, etc.) in usercfg.txt.
# Please refer to the README file for a description of the various
# configuration files on the boot partition.

ericjunkins, 2021-08-21:
@JHPHELAN i don't see these instructions at all in this:

https://github.com/nasa-jpl/osr-rover-code/blob/master/setup/rpi.md

JHPHELAN, 2021-08-21:
It's not in master.  It's in foxy-devel.  I need to add that to the issue.  I thought it would only appear under foxy-devel.

closed 2023-09-25 by Achllle