nasa-jpl/osr-rover-code #160: [bug report] screen.py status check functions: scaling incorrect

SallyTYT, 2022-09-29:
https://github.com/nasa-jpl/osr-rover-code/blob/master/ROS/led_screen/src/screen.py

The status data should not be scaled by a factor of 0.1

line 70 >> battery *= 0.1 
line 96 >> val = temperature[i][j] * 0.1
line 119 >> val = currents[i][j] * 0.1
line 140 >> val = currents[i][j] * 0.1

ericjunkins, 2022-09-29:
Can you give more context on why you think this is incorrect? I believe it is correct as is currently

SallyTYT, 2022-09-30:
@ericjunkins

If scaling by 0.1, the `out_msg[3], [5]~[12]` would always be zero,
[image]
And the LED screen would always be like this,
[image]

If removing the scaling, the `out_msg` would have valid values, It works just like this document describes https://github.com/nasa-jpl/osr-rover-code/blob/master/OsrScreen/README.md
[image]
And the LED screen would work well too,
[image]

closed 2023-09-25 by Achllle