nasa-jpl/osr-rover-code #169: Use C++ roboclaw library or switch to async python

Achllle, 2023-10-09:
A lot of time is spent waiting for communication with the roboclaws. Using an async python library to communicate with them or using threading in either language would alleviate this.

ericjunkins, 2023-10-09:
Sorry maybe I don't fully understand here, how would async or threading improve this? You always only have the ability to communicate to a single device at a time, so you couldn't multi thread that. Can you explain a little more?

Achllle, 2023-10-09:
Async or threading would simply mean that while the code is waiting for a response from the various IO devices, it can go and do other stuff such as respond to a callback or calculate odometry/state information.