nasa-jpl/osr-rover-code #171: Add support for duty control of drive motors

Achllle, 2023-10-09:
This is easy to add in code but we'll need to modify the command message or create a new one.

Right now the rover only supports velocity control. In some cases it could be better to command duty cycle to the drive motors as opposed to doing closed loop velocity control. This would likely mean less stress on the rover while driving, even though the motions are not as 'accurate' in some sense.

Achllle, 2023-11-06:
I changed my mind on this - using duty cycle just lowers speed, so you end up with less accurate speed control. The actual fix would be to change the ramping behavior of the speed control by changing acceleration settings.

ericjunkins, 2023-11-06:
I would make the case that during manual drive with a controller you might prefer open loop instead of closed loop speed control, and that closed loop is really only beneficial for autonomous functionality

Achllle, 2023-11-06:
True, true - I'll undo that change.

Achllle, 2024-03-30:
completed with #178

closed 2024-03-30 by Achllle