nasa-jpl/osr-rover-code #190: CMake Error during colcon build

murphcj, 2024-04-20:
See screenshots. When doing the colcon build step while setting up ROS, received the message "CMake Error at CMake Lists.txt:19 (find_package): By not providing "Findament_cmake.cmake" in CMAKE_MODULE_PATH this project has asked CMake to find a package configuration file provided by "ament_cmake", but CMake did not find one.

Could not find a package configuration file provided by "ament_cmake" with any of the following names:

ament_cmakeConfig.cmake
ament_cmake-config.cmake

Add the installation prefix of "ament_cmake" to CMAKE_PREFIX_PATH or set "ament_cmake_DIR" to a directory containing one of the above files. If "ament_cmake" provides a separate development package or SDK, be sure it has been installed" 

[image]
[image]

Achllle, 2024-04-21:
`ament` is the build system for ROS 2, which likely means ROS 2 isn't installed properly or not sourced properly. Someone else has asked [this question before on the `ros2` repository](https://github.com/ros2/ros2/issues/710). Try running `source /opt/ros/iron/setup.bash` in the same terminal and try running that line again. Let me know if that solves your issue.

P.S.: Just a note to make life easier: avoid posting screenshots of logs or instructions. You can refer to instructions from Github by adding a link to them (there's a link symbol when you hover over the header that you can copy) and you can copy paste logs using the "code" button in github to preserve formatting. This makes your issue searchable for others and allows copy pasting parts of the logs.

Achllle, 2024-05-09:
I was able to reproduce this by not sourcing, which is in the instructions. Since I didn't hear back from you and the remedy is in the documentation, I'm closing this. Feel free to comment if you still have questions or feedback!

closed 2024-05-09 by Achllle