nasa-jpl/osr-rover-code #195: ModuleNotFoundError: No module named 'RPi' in calibrate_servos.py

Achllle, 2024-08-23:
A slack member reported this issue when running `python3 calibrate_servos.py 0 150`. The fixed that worked for them was to `pip install RPi.GPIO`. It's not clear whether that should be the case for everyone and thus added to instructions.

The way to replicate would be to start with a fresh install or use a virtualenv or similar.

hariharanragothaman, 2025-01-23:
@Achllle  - Can I get this assigned to myself - I would like to fix this! Thanks in advance.

Achllle, 2025-01-23:
You're welcome to submit a PR. I don't assign issues to people since often the PR never comes through

hariharanragothaman, 2025-01-23:
Thanks! I will start actively working on it! 👍🏻

hariharanragothaman, 2025-01-29:
@Achllle  - Created a PR here: https://github.com/nasa-jpl/osr-rover-code/pull/203
Do let me know if I have to make any changes on my end.

Achllle, 2025-02-17:
closed by #187

closed 2025-02-17 by Achllle