nasa-jpl/osr-rover-code #207: Connected to /dev/serial1 but nothing happens

arambarricalvoj, 2025-02-12:
Firstly, I tried it on Ubuntu 24.04 LTS. After following [rpi.md](https://github.com/nasa-jpl/osr-rover-code/blob/master/setup/rpi.md) we got the Roboclaw controllers connected to `/dev/serial0` and did work and move.

However, I has to switch to Ubuntu 22.04 LTS and after following the same steps I get "Connected to /dev/serial1." and nothing else is output in the terminal, though it gets stuck.

How can I open or connect to `/dev/serial0` ?

Achllle, 2025-02-13:
Hi and welcome! Github issues are not the correct place to post questions unless you believe there's a specific problem with the code or instructions. Since many others have gotten this to work, you're most likely missing a step. Please join the Slack group (link in the main readme) and ask a question in the #troubleshooting channel to get fast support.

Since there are many possible causes for this issue, before you post make sure you've completed all steps in the instructions. Make sure you've enabled serial in `raspi-config` but not 'terminal over serial'. `/dev/serial1` is the correct device to connect to. Show the output for running the `roboclawtest.py` file as in the instructions. Finally make sure you've configured the [roboclaw parameters to connect to `/dev/serial1`](https://github.com/nasa-jpl/osr-rover-code/blob/d6077672c5b4ef6be4ce0f24d5340c6006173b4f/ROS/osr_bringup/config/roboclaw_params.yaml#L7). Hope that helps.

closed 2025-02-13 by Achllle

arambarricalvoj, 2025-02-13:
Hello Achille,
Solved! The problem was that 'terminal over serial' was activated. In fact,
on Ubuntu 24.04 LTS it did work with that configuration over /dev/serial0.

Let me know an appropriate forum in which I can post this information to
help other people.

Thank you,
Javier Arambarri Calvo.

Achllle, 2025-02-14:
Awesome. This is addressed in #187 which needs some validation.

Please join the Slack forum - we'd love to see your OSR whether in progress or completed!